artisan-roaster-scope/artisan #198: Roast properties: weight & volume to allow hundredths not tenths?

InvisibleOrdinary, 2018-05-09:
## Expected Behavior
To be able to enter x.xx when entering initial and ending weight… it would be really nice to be able to record the same accuracy for

## Actual Behavior

weight & volume only allow to enter to tenths of a unit

## Steps to Reproduce the Problem

## Specifications

  - Artisan Version: 1.3.0 
  - Artisan Build (number in brackets shown in the about box): (ca3c5f1)
  - Platform (Mac/Windows/Linux + OS version): Mac OS 10.13.4
  - Connected devices or roasting machine: phidgets 1048 

  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.

[180508_artisan-settings.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/1988499/180508_artisan-settings.aset.zip)

Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-05-11:
You should be able to enter begin and end weight to the tenths of a unit. See the attached screen shot using your settings file with weight unit set to lb.

[image]

What is exactly missing?

roasterdave, 2018-05-11:
I see the problem  The field input validator allows only one decimal place to be entered.   However, the result from a conversion or calculation is displayed with multiple decimal places.

closed 2018-05-18 by MAKOMO