artisan-roaster-scope/artisan #200: Bluetooth connection collection temperature.

WenatonGao, 2018-05-14:
## Expected Behavior
Bluetooth connection collection temperature.
[cs35_18-05-09_2309.alog于可意大利重.pdf](https://github.com/artisan-roaster-scope/artisan/files/1999447/cs35_18-05-09_2309.alog.pdf)

## Actual Behavior
Bluetooth connection failed.It is not a problem to test the bluetooth link with the serial debugging tool, but the Artisan serial connection is always unsuccessful, and the main performance is that the serial port cannot be opened. The timeout is set to 10, and occasionally the connection is successful, but the temperature update is more than 10 seconds.

## Steps to Reproduce the Problem

  1. Bluetooth connection matched successfully.
  2. Artisan->Config->TC4->OK Show the ports configuration window more than 10 seconds later.
  3. Select the bluetooth virtual serial port (the response is very slow after clicking)-> is set correctly ->OK.
  4.After clicking ON, "Serial Exception: Unable to open Serial port"

## Specifications

  - Artisan Version: V1.3
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):Win10
  - Connected devices or roasting machine:Homemade temperature acquisition board, microcontroller STM32F103, and its serial port connection bluetooth 2.0 module (hc-06)，The data interface is the same as TC4.
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-09-05:
Do you have any suggestion on how to improve the serial communication of Artisan? Please submit it as Pull Request.

MAKOMO, 2018-09-11:
I close this for now as Bluetooth is not part of the supported protocols of Artisan

closed 2018-09-11 by MAKOMO

WenatonGao, 2018-09-11:
My Bluetooth can be used in the V1.1 version. There is no other solution.