artisan-roaster-scope/artisan #202: Highlight FCs-to-FCe and SCs+ in the other themes like the default does as a shaded area

Tonefish, 2018-05-17:
Enhancement
## Expected Behavior
FCs-to-FCe and SCs+ highlighted in profile plot for themes
## Actual Behavior
FCs-to-FCe and SCs+ not highlighted in profile plot for themes except default

roasterdave, 2018-05-17:
This is the intended behavior.  However, you can turn those highlights on in any theme and save as a personalized theme.  To turn on the highlights go to Config>Phases and tick Watermarks.  Make any other color changes you want from Config>Colors.  Once you have it the way you like, save your theme using Config>Themes>User>Save Theme.  Re-load your personalized theme any time you want.

Tonefish, 2018-05-17:
Wonderful! Thank You!!

Should the defaults for each theme be the same though or possibly carry over those settings as you switch themes? What made me think they were missing is that they disappeared when I changed themes. Thank you all for what you do here.

roasterdave, 2018-05-18:
I understand your suggestion.  These highlights are a part of the color set so the choice was made to include the Watermarks setting in the theme.  The idea behind the additional Artisan themes was to show what is possible and to provide some starting points for users to create their own themes.  We can include some notes about the themes in the documentation so that this is clear. 

Thanks for your comments on this!

closed 2018-09-05 by MAKOMO