artisan-roaster-scope/artisan #204: Assignable Keyboard Shortcuts

micahvdm, 2018-05-18:
Hi,

Is there a possibility of getting assignable keyboard shortcuts to the buttons for charge, dry end, etc...
It would be nice to have this feature instead of having to use a mouse to click them every time.
I think it would bring more focus to the roasting process.

MAKOMO, 2018-05-18:
There is a keyboard navigation mode. Try to press ENTER/RETURN during a roast. Now you can use the cursor keys to select one of the buttons (usually the correct next in the row is already selected). Use the SPACE key to "press" it.

MAKOMO, 2018-05-18:
Difficult. Most keys have already some assignment (check the keyboard shortcuts entry under the Help menu). Most likely keyboard shortcuts will not be assignable in the near future. Sorry.

closed 2018-05-18 by MAKOMO

elmills, 2018-12-03:
I know that this issue is closed and is completely at your discretion, but I would appreciate if it was reconsidered.  I am looking at using an Elgato streamdeck and/or other "keyboard" like controllers to "hit" buttons.  It would be nice to be able to specify a custom keyboard shortcut for different actions.  Then I can map something rare/unexpected, like ctl+123 to charge, ctl+124 to dry end etc.  

That way I have very clearly labeled buttons on the control and the can quickly hit them as I am about to drop, turn on a heater, etc.  Long term I a moving to fully automated control within Artisan, but am not to that point now.

[image]

InvisibleOrdinary, 2018-12-06:
+1 for this… having something like this would be enormously helpful in streamlining the entering of data…

elmills, 2019-01-02:
@MAKOMO I would be interested in "sponsoring" this feature depending on the cost.  Not sure that is possible, but would be willing to discuss.  Thanks.

MAKOMO, 2026-01-29:
By accident I found this old issue. Note that Artisan can be fully controlled by a Streamdeck for some years by now.

https://www.home-barista.com/roasting/keypad-control-artisan-t82836.html