artisan-roaster-scope/artisan #205: Enable PID ON / PID OFF Alarm Actions for Arduino/TC4

James-NZ, 2018-05-18:
## Expected Behavior
Alarm actions PID ON and PID OFF should enable / disable hardware PID on Arduino/TC4 platform. RampSoak ON / OFF may also be affected.

## Actual Behavior
PID state on Aruidno/TC4 does not change when alarm activates.

## Steps to Reproduce the Problem
Alarm handlers only work if aw.qmc.Controlbuttonflag is set (SW only?). aw.qmc.Controlbuttonflag case is handled in pidOn / pidOff functions so this condition may be able to removed from the alarm handlers to keep them generalized for all PID types.

main.py : 2802-2827
            elif self.alarmaction[alarmnumber] == 17:
                # RampSoak ON
                if aw.qmc.device == 0 and aw.fujipid: # FUJI PID
                    aw.fujipid.setrampsoak(1)
                elif (aw.pidcontrol and aw.qmc.Controlbuttonflag): # internal or external MODBUS PID control
                    aw.pidcontrol.svMode = 1
                    aw.pidcontrol.pidOn()
            elif self.alarmaction[alarmnumber] == 18:
                # RampSoak OFF
                if aw.qmc.device == 0 and aw.fujipid: # FUJI PID
                    aw.fujipid.setrampsoak(0)
                elif (aw.pidcontrol and aw.qmc.Controlbuttonflag):  # internal or external MODBUS PID control
                    aw.pidcontrol.svMode = 0
                    aw.pidcontrol.pidOff()
            elif self.alarmaction[alarmnumber] == 19:
                # PID ON
                if aw.qmc.device == 0 and aw.fujipid: # FUJI PID
                    aw.fujipid.setONOFFstandby(0)
                elif (aw.pidcontrol and aw.qmc.Controlbuttonflag): # internal or external MODBUS PID control
                    aw.pidcontrol.pidOn()
            elif self.alarmaction[alarmnumber] == 20:
                # PID OFF
                if aw.qmc.device == 0 and aw.fujipid: # FUJI PID
                    aw.fujipid.setONOFFstandby(1)
                elif (aw.pidcontrol and aw.qmc.Controlbuttonflag): # internal or external MODBUS PID control
                    aw.pidcontrol.pidOff()

Might be worth checking if this pattern occurs elsewhere in the code (e.g. key press handlers: ln14861, ln14874, ln14881?).

## Specifications
  - Artisan Version: 1.3.0
  - Artisan Build (number in brackets shown in the about box): ca3c5f16
  - Platform (Mac/Windows/Linux + OS version): Win64
  - Connected devices or roasting machine: Arduino with TC4 PID firmware

MAKOMO, 2018-05-19:
Dear James, thanks for this detailed report. Instead of changing the pre-conditions in the cases you indicate, wouldn't it be more consistent to just enable the PID features and show the "Control" button if the "Control" flag is ticked also for the case the TC4 PID firmware is selected? That way a user could run the TC4 PID firmware and still deactivate the PID features (eg. for testing something) with one tick. It seems that if the "Control" flag is ticked the PID alarm actions work as expected as are the keyboard shortcuts. Do I overlook something?

James-NZ, 2018-05-19:
Hi Marko, yes that sounds good as it doesn't look like there are any side effects of having the Control Flag ticked too, as the critical code is guarded with if/elif statements to catch the TC4 PID actions first.

As you suggest, for consistency you could then hide the Control Button if the Control Flag is unticked for the TC4 PID as well:

line 12939-12947:
```
    # decides on visibility of the Control button based on the selected devices and configuration
...
        elif aw.qmc.device == 19 and aw.qmc.PIDbuttonflag: # ARDUINOTC4
            res = True
```
This will then also match the description here:
https://artisan-roasterscope.blogspot.de/2016/11/pid-control.html

>**Fuji PXG/PXR, Delta DTA and Arduino TC4**
>Artisan supports the configuration and operation of three hardware PIDs in a specific way. The Fuji PXG, the Fuji PXR and the Delta DTA. If any of those is configured as main device (menu Config >> Devices). If one ticks the "Control" flag in the device configuration dialog, an extra "Control" button appears on the main window next to the "START" button.

That just leaves the FUJI PID to have a think about, as it is handled irrespective of the Control Flag :)

MAKOMO, 2018-05-20:
Thanks for your comment. Hope I did it correct now in v1.3.1

closed 2018-05-20 by MAKOMO