artisan-roaster-scope/artisan #206: Hash Marks on Sliders

JLBennettco, 2018-05-20:
## Expected Behavior
Hash Marks

## Actual Behavior
They used to be there, now they're not. I miss them!

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version:
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-05-21:
Which hash marks!? What do you refer to by "on sliders"? Could you upload a screenshot for clarification?

MAKOMO, 2018-05-21:
Wait a second, I think I know what you are talking about. Those position annotations (tick marks) that the Artisan sliders had until v1.3 where we introduced the color themes. I am sorry to say that its seems currently close to impossible to have those tick marks nicely rendered on all platforms and supporting color themes.

MAKOMO, 2018-07-07:
Adding back those hashmarks seems to be a complex task under the new color themes for some internal technical issues. For now we decided to spend our resources on other things as the current slider value is precisely shown at any time in the sliders LCDs. Those marks do not really add much information over this. Sorry.

JLBennettco, 2018-07-07:
Understood. Thanks for the work.

closed 2018-07-13 by MAKOMO