artisan-roaster-scope/artisan #207: v1.3.1 complains about phidget v22

edgndg, 2018-05-21:
## Expected Behavior
Running linux mint 18.2 on x64 laptop.  Version 1.2 has no issues.  Program finds installed phidgets drivers.

## Actual Behavior

Upgraded from 1.2 which works fine.  On 1.3.1 get the following error:  0:01:21.650 Exception: Phidget v22 driver not installed.

The driver is installed.  All paths are correct.  Have uninstalled and reinstalled both drivers and program to no avail.

## Steps to Reproduce the Problem

  1. start the program
  2. start recording
  3. error message

## Specifications

  - Artisan Version:
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-05-21:
Does the Phidgets Control Panel attach correctly to your Phidgets? Did you verify that the HelloWorld program (see https://www.phidgets.com/docs/OS_-_Linux)  runs correctly on your setup? Are the UDEV rules (see last section of https://www.phidgets.com/docs/OS_-_Linux) installed correctly? Is the correct Phidgets device selected within the Device setup of Artisan?

edgndg, 2018-05-21:
Hi -- thanks for quick response.

In reverse order: yes the device is selected, yes I added the UDev rule (but to back test I have run as both regular user and root), I did not do the hello world as it worked out of the box on artisan 1.2 (and continues to do so when I downgrade from 1.3).

I did some additional poking yesterday -- I had added an ld.so.conf entry and run ldconfig when I installed 1.2 b/c phidgets drivers install to /usr/local/lib by default.  Adding symlinks to /usr/lib seems to resolve the issue in 1.3.  So it works for me, but something seems odd about where it is looking for the shared libraries.

MAKOMO, 2018-09-11:
It seems that Phidgets fixed this on their side in the 1.0.0.20180827 release saying "Fixed Python bug not being able to find library on Linux".

closed 2018-09-11 by MAKOMO