artisan-roaster-scope/artisan #208: Background profile title appears to be a fixed black color rendering it unreadable in some themes

Tonefish, 2018-05-22:
## Expected Behavior
Background profile title text color would change with the themes axes text color and title text color

## Actual Behavior
Background profile title text color appears to be fixed to a black color (couldn't find it in config colors)

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version:
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

roasterdave, 2018-05-22:
Thanks for the report.  In version 1.3.1 (all builds) the background title color is now set to the same color as "Title".  Let us know if this is a problem for you, or not, in version 1.3.1.

If it is still a problem, and for all issue reports please include the Artisan version, build number and the platform you are using when reporting issues.  It makes helping much more efficient.

Tonefish, 2018-05-22:
Thanks Dave.  Just noticed that and came back to say never mind.  Thank You Again!!

closed 2018-05-23 by MAKOMO