artisan-roaster-scope/artisan #209: set settings for beginning sequential batches

InvisibleOrdinary, 2018-05-23:
## Expected Behavior
When roasting several batches, it would be great to have a way to return to original settings (sliders at x & y, etc.) automatically when starting the next roast. 

## Actual Behavior
no ability (that I see) to do this

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version:1.3.0 
  - Artisan Build (number in brackets shown in the about box):(ca3c5f1) 
  - Platform (Mac/Windows/Linux + OS version): Mac OS 10.13.4 (17E202)
  - Connected devices or roasting machine: Proaster 10K w/ phidgets 1048 (2 probes)
[180522_artisan-settings.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/2029498/180522_artisan-settings.aset.zip)

  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-07-07:
Good idea. We will work something out on tackling "Issue #185 Programable pre-heating" and roast schedules.

MAKOMO, 2025-08-13:
Most parameters can be set by Artisan Commands which can be triggered by Alarms for example on turning Artisan ON for the next batch to have defined conditions at the start of a roast. Seems to be resolved.

closed 2025-08-13 by MAKOMO