artisan-roaster-scope/artisan #212: Events buttons - add line below instead of above seems to be more usable

Tonefish, 2018-05-23:
## Expected Behavior
Events buttons add line adds below the present line

## Actual Behavior
Events buttons add line adds above the present line

Thank You for adding this feature but I believe a slight modification would make it much more useful w/o creating a different problem. The feature is excellent for modifying a list already made, but the present form makes it a real nuisance to build a list in the first place because you have to start at the bottom of the list and build upward.  I believe adding the line below serves both purposes (starting a list and modifying a list) very well. Thank You Again!

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version: 1.3.1
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-05-23:
Stupid me! My idea was to use the "Add" button for both, insert and add action appending at the end. The insert action would only trigger if a row would be selected. Fine if you only could deselect the current selected row in any way. Sorry. I just added a proper "Insert" button that inserts above the selected row as it is now, but the "Add" button will always add an element at the very end.

Tonefish, 2018-05-23:
If you only wanted one button I think adding the line below would work for both with just the add button.  Thank you for a great work of software. Jim

MAKOMO, 2018-05-24:
Thanks Jim for raising the issue and this suggestion. However, adding below the selected row would not allow to add a row before the very first one. Thus I decided for the extra button now, also as this aligns better with the "Insert" button I just added in v1.3.1 to the alarm list with the very same functionality. The current v1.4 builds on bintray <https://bintray.com/artisan/artisan-artifacts/artisan/master#files> already contain this change and are save to use.

closed 2018-07-07 by MAKOMO