artisan-roaster-scope/artisan #214: Events - Sorting

chuckcoffee, 2018-05-27:
## Expected Behavior
Can you add feature of being able to sort the events. One could use one of the columns ie documentation as your sort column or you could add a new column just for sorting

## Actual Behavior

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version: 1.3.1
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version): Win 7
  - Connected devices or roasting machine: Huky 
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-05-30:
Could you be a bit more specific on what you want to have sorted? The event table under Roast >> Properties, 3rd tab events is sorted by time (if not as you added manually some entries you can press "Order"). I cannot see a "documentation" column here, maybe you refer to "Description". What is the purpose of an extra sort column? Please give us a short example to help us understand your suggestion. Thanks!

chuckcoffee, 2018-05-30:
Marko

Sorry guess my wording was off. I meant the buttons

So some way to sort the
 Fan sequnetially, gas sequentially

chuckcoffee, 2018-05-31:
[image]

chuckcoffee, 2018-05-31:
So if the order of the buttons is not sequential I believe you currently need to delete all of then and then reenter in the order you want. With a sort feature this could be avoided

dabrjn, 2018-10-06:
I too would like to be able to reorder the buttons, but just the capability to move the rows up or down would work great. A sort feature would like sort all the fan buttons by type and value wouldn't work for me while a move feature would.

MAKOMO, 2019-04-19:
Resolved by PR #345 (thanks Phil!)

closed 2019-04-19 by MAKOMO