artisan-roaster-scope/artisan #215: Artisan v. 1.3.1.1 S7 Communication Error on Giesen WP15A with HMI 5.5

BekanTeg, 2018-06-06:
## Expected Behavior

Artisan v. 1.3.1.1 normaly connet to Giesen WP15A with **HMI 5.45 5.5 6.0 PLC 5.45 5.5 6.0**

## Actual Behavior

Artisan v. 1.3.1 (1) can't connect to Giesen WP15A with  **S7 Communication Error** with HMI 5.5 PLC 6.0 or 5.5 and HMI 5.45 PLC 5.45
Artisan v. 1.3.1 (1) connect normaly to Giesen WP15A with **HMI 6.0 PLC 6.0**

On **Artisan 1.3.0 (ca3c5f16)** connect with Giesen WP15A with **HMI 5.45 5.5 6.0 PLC 5.45 5.5 6.0** **is OK.**  

## Steps to Reproduce the Problem

  1. Connect to Giesen Coffee IT services and updating PLC to 6.0 (HWI we can't update is hardware box)
  2. Reinstal the Artisan 1.3.1 (1) program don't help
  3. Loading artisan settings from 1.3.0 (ca3c5f16) to 1.3.1 (1) don't help 

## Specifications

  - Artisan Version: 1.3.1
  - Artisan Build (number in brackets shown in the about box): 1.3.1  (1) 
  - Platform (Mac/Windows/Linux + OS version): Windows 10 64 bit
  - Connected devices or roasting machine: HP Probook 4720s
[artisan-settings.zip](https://github.com/artisan-roaster-scope/artisan/files/2075287/artisan-settings.zip)

Thank you for you help!
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-06-13:
Giesen/S7 support did not change between v1.3.0 and v1.3.1 of Artisan. Are you sure that v1.3.0 connects successfully to that "HMI 5.45 5.5 6.0 PLC 5.45 5.5 6.0" machine while Artisan v1.3.1 does not, using the very same settings?

lookingafar, 2018-06-13:
I can confirm this. Artisan 1.3.0 connects to HMI 5.5 and PLC 5.5 and 6.00 successfully, while Artisan 1.3.1 and 1.4.0 does not. The error message is "S7 communication error"

BekanTeg, 2018-06-13:
Lookingafar, thank you for you support!

MAKOMO, 2018-06-14:
Dear BekanTeg and Lookingafar,

I compared the S7 code used by this Giesen connection between v1.3.0 and v1.3.1/trunk and couldn't see any essential differences. The cause of the communication issue might be related to some timing differences. To work this out I depend on your support as I do not have access to any Giesen hardware.

I just did add slightly better error logging to the S7 code to understand better where it fails and also changed the timing during connect slightly. Could you please run the current builds on bintray that have just been generated

https://bintray.com/artisan/artisan-artifacts/artisan/master#files

and send me your error and message logs (available under menu Help). Additionally, it would be good to know on which OS you are running the test (Windows/Linux/Mac) and what exactly happens (does Artisan ever connect to the machine successfully "S7 Connected", does it ever retrieves some readings, does the slider control for fan/burner works?).

Thanks for your support,
Marko

MAKOMO, 2018-06-14:
Upps. Just see that the builds are broken. Wait a minute, I try to fix them...

lookingafar, 2018-06-14:
Thank you! I will try the builds today and send you the logs

MAKOMO, 2018-06-14:
Ok. Only the Mac OS X build was broken. Fixed that one. New builds are now online on bintray. You can start testing.

MAKOMO, 2018-06-14:
Thanks lookingafar for your support on this! It is a tricky thing. Hope will be able to resolve this soon.

lookingafar, 2018-06-14:
I've tried the last build.

Here's the message log.

-----------------------------
24 13:28:34.598 Scope stopped

23 13:28:34.438 S7 Communication Error readInt: name 'util' is not defined

........(same messages here)

6 13:28:17.262 S7 Communication Error readInt: name 'util' is not defined

5 13:28:16.252 S7 Communication Error readInt: name 'util' is not defined

4 13:28:15.244 S7 Communication Error readInt: name 'util' is not defined

3 13:28:14.789 S7 Communication Error readInt: name 'util' is not defined

2 13:28:14.194 Scope monitoring...

1 13:28:14.156 Scope has been reset
-------------------------------------

The error log says:
-----------------------------------------
1 13:29:45.295 S7 Communication Error readInt: name 'util' is not defined
------------------------------------------
(there are lots of exact same messages, I haven't found any different messages)

I've tried it on Windows 10 so far.
Sliders don't work, there are no readings. Here is the screenshot.

[image]

lookingafar, 2018-06-14:
Just tried it on linux. Clean install (1.4.0), loaded default setting for Giesen (Congif > Machine > Giesen), pressed "On". Same thing. The error message is the same. I haven't tried loading settings from 1.3.0 on linux, nor have I tried to install 1.3.0 on linux to see if it works. I'll do it if you need.

MAKOMO, 2018-06-14:
Thanks for your quick test. It revealed that the Windows and Linux builds of Artisan v1.3.1 had a broken the S7/Giesen thing (remaining undiscovered as I am mostly testing on Mac OS X). The builds now online on bintray should fix this. At least Artisan under Windows/Linux should now start communicating with the machine. Let's see how successful that is. Please report! Do you also have access to PLC v7 versions? I was told that there might be an issue too. Thanks for your support, Marko

lookingafar, 2018-06-14:
I don't have access to PLC v7. We have a Giesen W15A and guys from Giesen recently updated our PLC from 5.5 to 6.00 (so it seems that it's the last version they use right now for our model). Maybe some newer versions of their roasters have PLC 7.0. I'll ask fellow roasters who use Artisan and Giesen.

MAKOMO, 2018-06-14:
@lookingafar Thanks for testing! I'll wait for a confirmation from @BekanTeg before I close this one. Expect this to go into the next release version.

BekanTeg, 2018-06-14:
@MAKOMO, @lookingafar thank you for you help! Nice job!

BekanTeg, 2018-06-15:
I think we can close this issue :)

closed 2018-06-15 by BekanTeg