artisan-roaster-scope/artisan #219: Artisan .alog format, poor JSON implementation

digipifini, 2018-06-18:
## Expected Behavior
Should conform to the standardized JSON format, allowing for editing using multitude of online tools.

## Actual Behavior
JSON formatting with many errors when checking using JSON syntax tools.

## Steps to Reproduce the Problem

  1. Create .alog file and save
  2. Upload to any online syntax tool
  3. Allow errors to pile up

I'm not quite sure if anyone else has noticed these issues.  After first reviewing the structure of the .alog files I started to notice it is a somewhat problematic implementation of JSON, with many basic syntax rules overlooked.  If this is to be considered opensource we might want to follow proper structuring instead of created yet another proprietary standard making it more difficult for newcomers to adapt to.

rpaulo, 2018-06-18:
The alog files are not valid JSON.  They are essentially the print out of python data structures.

MAKOMO, 2018-06-18:
Use menu File >> Export to >> Artisan JSON to generate valid JSON structures from .alog profiles.

MAKOMO, 2018-06-18:
samplinginterval should never be None btw. How did you generate that Artisan profile? Could you append this profile for testing?

digipifini, 2018-06-18:
Thanks for the quick feedback guys, very much appreciated as I am fairly new here.  @rpaulo thanks for the clarification, I assumed since it appeared to be so close to a json structure, albeit with the errors mentioned.  Could it be taken as a possible direction for future versions of the .alog files to possibly become a json format? As I mentioned, open source and common/universal file types help make the information more open between applications, it would also eliminate having to export to a format which is really in close in syntax.  Just my two cents, the fewer file types the better.

@MAKOMO I saw the JSON export feature when I initially started trying out the artisan software, and as stated it appeared so similar, but thank you for the correction.  I create a profile using the designer feature and then imported after saving, possibly this is the reason for the None appearing in the stated value.

closed 2018-07-07 by MAKOMO