artisan-roaster-scope/artisan #227: Event Alarm Action

arkenamun, 2018-07-09:
Feature request.

I would like to request that Event is added to the Action options in alarms.

Maybe it could be called "Custom Event" to differentiate it from the existing "Event Button" option, to avoid confusion.

Additionaly it would be useful to write the event description in the alarm description field.

Justification for request: Many roasters keep records of and compare the time that certain temperatures are reached during the roast, this feature would allow thie presentation of these data on the profile to be simply automated with alarms.

MAKOMO, 2018-07-12:
Rejected. Define an event button with the intended action and make it invisible. Use the "Event Button" alarm action to trigger that button.

closed 2018-07-12 by MAKOMO

arkenamun, 2018-07-16:
Hey Marko,

Thanks for that explanation, I had not considered using an event button. Brilliant! That works perfectly in conjunction with the 'Mini Editor' to see on screen at what time (mm:ss) a defined temperature is reached during roasting. As I didn't want to trigger any external event at these points, aside from recording the point in the roast, I used an event type of +/- slider with value of 0. The button description appears in the Mini Editor and roast properties event description, which is perfect.

Thanks again!