artisan-roaster-scope/artisan #228: Export/import format mismatch for Probat pilot ver. 1.4.*

vold029, 2018-07-10:
## Expected Behavior
1) Exported with menu File->Export->Probat Pilot... file can be successfully opened in Probat Pilot Shop roaster software.

2) Imported with menu File->Export->Probat Pilot... file displayed as profile curves in Artisan.

## Actual Behavior
1) Get error message when importing file in Probat Pilot Shop roaster software. File exported from Artisan with menu command: File->Export->Probat Pilot... 

2) Artisan show clean roaster scope arter menu command File->Export->Probat Pilot... 
"Scope has been reset" message displayed in upper left corner. 

## Steps to Reproduce the Problem

EXPORT: 
  1. Record profile in Artisan or create designer
  2. Export with menu File->Export->Probat Pilot... to file "test_1.xml"
  3. Open Pilot Roaster Shop software version 1.4.1.0
  4. Choose Recipes->Import->"test_1.xml"
  5. Get error message: "Error during file import! Error while reading recipe test_1.xml: Input string was not in a correct format."

Profile file format has changed in version  1.4.+ Pilot Roaster Shop software.

IMPORT:
  1. Choose: File->Export->Probat Pilot... 
  2. Select file for import (attached as example)
  3. Get info message: "Scope has been reset" message displayed in upper left corner. Roaster scope stay empty.

[artisan-settings.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/2180330/artisan-settings.aset.txt)

Exapmle of new format file attached.
[100% _0.xml.txt](https://github.com/artisan-roaster-scope/artisan/files/2180277/100._0.xml.txt)

## Specifications

  - Artisan Version: 1.3.1
  - Artisan Build (number in brackets shown in the about box): 1
  - Platform (Mac/Windows/Linux + OS version): Windows 7/64
  - Connected devices or roasting machine: Probatone 12.
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-07-12:
I updated the Pilot import/export taking your Pilot xml file as guide. This is half-blind as I lack a proper specification of the Pilots file format as well access to the Pilot software. A request to Probat is send. You can check the current v1.4 beta builds on bintray https://bintray.com/artisan/artisan-artifacts/artisan/master#files if those work for you.

[image]

vold029, 2018-07-12:
Marko,

thank you for fast response and provided solution.

Please, stay connected and inform me about all Artisan updates .

I still keep my eyes on all news.

Best regards,

Vladimir Klybik

MAKOMO, 2018-07-13:
Could you test this on your side and verify that import and export now work as expected? Can I close this issue?

vold029, 2018-07-13:
Today I'm done clean install of Artisan 1.4.0, do factory reset configuration after install and get this issues after importing attached test file:
- BT seems to be imported correctly
- Burner level imported not as default "burner" event, but as some type of external event (an also "Extra 2" event). Then I try to switch off display "burner" graph by clearing appropiate check box, it stay on display. See on attached screen_1.png, marked as "1)". Also, you can see this on screen_2.png, where I switch lunguage to "Russian".
- Burner level duplicated as default "Drum" event. See mark "2)" on attached screenshots. As I know, some new Probatone roasters equiped with drum speed controller, but not in our case and I can't see drum speed info in source Probat Pilot xml file. 
- "drum" and "burner" events graphs shifted on display vertically one from one. Although, one time (first after Artisan install) I can see two graphs matched exactly, but can't repeat this after.
[artisan-settings.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/2192378/artisan-settings.aset.txt)
[image]
[image]

vold029, 2018-07-13:
> Dear Vlad,
>
> thanks for testing. Some comments on your findings. The burner level is
> imported in two ways. First a regular data curve (the black line marked
> correctly "Burner" in your screenshots) by using one extra device curve.
> Second, the pure burner changes are import also and generate Artisan
> Events. Artisan supports 4 of those event channels (disjoint from the
> standard temperature curves) that can be rename by the user. The Pilot
> incorrectly imported the events using the 2nd Event channel instead of the
> last one that is by default named "Burner". I just fixed that one (not
> commited yet).
>
> The "Burner" curve can be hidden from within the Extra Devices tab (menu
> Config >> Devices, 2nd tab), where you can also assign a different label.
> The "Burner" (4) events (currently in your setup the "Drum" (2) events) can
> be hidden by deactivating the corresponding flags in the Events dialog (and
> here you can change the name of the events channel as you already
> discovered).
>
> By default, events are drawn in their own axis system independent of the
> temperature axis used for the curves. This explains the discrepancy between
> those two elements in your screen shot. If you tick "snap" in the Events
> dialog, Event axis are aligned with the temperature axis.
>
> By now I have received some sample files from Probat for testing and also
> could download the current version of the Shop Pilot. I will test
> import/export with these and fix errors if I encounter them.
>
> Thanks again for your detailed report!
> Marko

Dear Marko,

thanks a lot for detailed explanation. Now all behaviour are clear for me.

Please, inform me on your next improvements in import/export Probat Pilot
files.

Best regards,
Vladimir

MAKOMO, 2018-07-14:
Dear Vlad, I applied a last change such that also profiles generated by the designer and exported to Probat Pilot format can be imported successful. Note that recipe files have to be named <Name>_<Index>.xml (like eg. "Test_1.xml") to be accepted by pilot. I added a corresponding warning popup.

closed 2018-07-14 by MAKOMO