artisan-roaster-scope/artisan #229: Artisan 1.3.1 automatically triggering SCb/SCe/Dump…

InvisibleOrdinary, 2018-07-11:
## Expected Behavior

The only states that are auto triggered are Turn & 'DRY END', in addition it auto aligns 'CHARGE'... everything after that is manual

## Actual Behavior

I cannot tell how this is happening, it's not consistent, so sometimes it auto triggers everything after 'FC START' (up to 'DROP'), sometimes it auto triggers everything after 'FC END' (up to 'DROP'), and sometimes it works as expected where everything after 'DRY END' is manual.  

The roast I uploaded is one where it auto-triggered the states after 'FC START'. I don't know when it did this.

To undo, I have to click each state a couple times to toggle them on/off/on/off until they are reset… this doesn't seem to be consistent, either, though every time, I am in the middle of First Crack, or towards the end of it, and am in a hurry to correct it.

It's never triggered 'COOL END'

## Steps to Reproduce the Problem

I don't see a pattern here.

## Specifications

  - Artisan Version: 1.3.1 (but this has been doing this for a few versions, back to 1.2.x I believe)
  - Artisan Build (number in brackets shown in the about box): (1)
  - Platform (Mac/Windows/Linux + OS version): Mac OS
  - Connected devices or roasting machine: Proster + Phidgets 1048 (2 thermocouples)
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.
[180710.zip](https://github.com/artisan-roaster-scope/artisan/files/2182901/180710.zip)

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

InvisibleOrdinary, 2018-07-11:
Hmm, it was in the folder attached to the ticket… anyway, attached separately. did you get the alog file?

[180710_artisan-settings.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/2185435/180710_artisan-settings.aset.zip)

MAKOMO, 2018-07-13:
I don't see anything specific in your setup that could cause this as you not even have any alarms defined. Those events are exclusively bound to the keys. As this cannot be reproduced and was not reported by anybody else I assume a local installation issue, maybe some other intelligent software like "Siri" is causing this, I close this issue.

closed 2018-07-13 by MAKOMO