artisan-roaster-scope/artisan #233: Feedback based automation request

lookingafar, 2018-08-05:
**This is a feature request.**

I am not sure if this is even possible to implement well, but let me know what you think.

Right now the automation process for supported roasters looks like this:
1) Record events
2) Trigger all the recorded events depending on time or BT/ET

The software doesn't get any feedback from the roaster, i.e. the ROR, the burner level, etc and certainly doesn't calculate anything, which is the definition of a system without feedback. And this kind of system has certain flaws, the main one being consistency. I.e. if the amount of energy inside the roaster is different next time, then the roast will be completely different, even though all the events will be triggered on time and at certain defined temperature points.

Now, the system with feedback can analyze the deviation between the needed result (like a certain temperature at a certain point of time) and the result we are getting or about to get. In other words, it can calculate how it can adjust the recorded profile (or even make it's own) based on the certain parameters the user has put into the system before the roast.

Let me give you an example:

Let's say there is a roasting profile. We roast 10 kg of coffee and we want to hit 140 degrees C at 5:00, 193 degrees C (which is FC) at 9:30 and then drop at 207 degrees C at 11;30 with the ROR about 4 degrees per minute.
So we put all these parameters into the profile, preheat the roaster, set the burner to 90%, drop the beans and hit "Charge". Now, for example, there is a background reference roast loaded with the same parameters (imo, without a preloaded profile it would be very difficult to calculate how to meet all the criteria). After a minute Artisan sees that the roast is 10 seconds behind from the reference roast, so it puts the burner to 100%.
Now we hit 140 at 5:00 or so and the roast proceeds with all the prerecorded events. At some point (at 160 degrees C, for example) Artisan calculates, that the roast is 10 seconds ahead from the reference. So, if the burner at this point is 80%, it puts it to 70% and adjusts all the other events as well (if the next recorded burner level is 70%, then it will become 60%). And so on...

----------
So, imo, the easiest way to start implementing this is to make it possible to create alarms based on the deviation from the background profile. I.e. if the deviation is +10 seconds or more at a certain time/temperature point, reduce the burner by 10%, if the deviation is -10 seconds or less, then increase the burner by 10%.
Imo, being able to put a certain time/temperature point as a second condition is quite important, since a 10-second difference can snowball to a 30-second difference after a couple of minutes, on the other hand a 10-second difference 15 seconds before the drop is not important.
I am not sure what to do with the rest of the events (burner points) in case the burner level changes by the alarm. There probably should be an option to adjust the next N burner changes to a certain % (I hope you see what I mean here)
So, the alarm conditions woud look like this:
1) If from 1:00 till 11:00 the time deviation from the reference roast is between -10 and -20 seconds, increase the current burner level by 10% and increase the burner level of the next N events by 10%
2) If from 1:00 till 11:00 the time deviation from the reference roast is between -21 and -40 seconds, increase the current burner level by 20% and increase the burner level of the next N events by 15%
Ideally, there would be an option to return the modified burner levels back to their original values when the deviation goes back to -5 seconds, for example.
3) If from 200 degr. C to 203 degr. C at least 6 minutes after the start the time deviation is between -20 and -40 seconds, set the burner to 100% for 10 seconds, then return it to the prerecorded value (proceed with the prerecorded profile)
-----------

With time, though, it could be developed into a complete automation system which constantly adjusts the machine (the burner level, the air dumper, etc), based on how the roast is developing and how close it is to the reference roast. Or the one that can create the whole roasting profile (and execute it) based on certain criteria. 

We roast on Giesen and we could help you with everything we can to impement this properly.

Thank you.
Marat.

MAKOMO, 2018-09-07:
Doesn't the PID loop of Artisan provides what you are looking for? See https://artisan-roasterscope.blogspot.com/2016/11/pid-control.html

MAKOMO, 2018-10-20:
Closing this as no further feedback was received

closed 2018-10-20 by MAKOMO