artisan-roaster-scope/artisan #235: Phase LCDs

Peaberrys, 2018-08-16:
Hi, is it possible to have the FCs Phase LCD time, percentage & temp modes shown all at once after selecting the FC start button? I find toggling through them fiddly on the MacPro track pad. This function would be great (if it doesn't already exist) :) 

## Specifications

  - Artisan Version: 1.3.1 
  - Artisan Build (number in brackets shown in the about box): 1
  - Platform (Mac/Windows/Linux + OS version): Mac 10.13.6
  - Connected devices or roasting machine: HUB0000_0 - VINT Hub Phidget, TMP1101_0 - 4x Thermocouple Phidget, 1139_0 - Differential Air Pressure Sensor 100 kPa
 
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file. 
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

[artisan-settings.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/2292249/artisan-settings.aset.txt)

MAKOMO, 2018-08-18:
Thanks for the suggestion. This mode was just added and will be part of Artisan v1.4

closed 2018-08-18 by MAKOMO

Peaberrys, 2018-08-18:
Thanks Marko. Is there any plans to be able to connect to the thermocouples and not have to disconnect the phidgets etc to save profiles or change some setting?

Kind regards,

Eric Smith

Closed #235<https://github.com/artisan-roaster-scope/artisan/issues/235>.

Peaberrys, 2018-09-02:
Hi Marko,

When are you looking at releasing software update 1.4?

Kind regards,

Eric

No Eric. For various reasons.