artisan-roaster-scope/artisan #238: Temp axis increase?

RoasterJ, 2018-08-22:
Is it possible to get the temp axis to read above 1000 degrees F? I have my it set to 1300 and as soon as my temp line hits 999 my ET reading goes to -1.

## Specifications

  - Artisan Version:1.3.1
  - Artisan Build (number in brackets shown in the about box):1
  - Platform (Mac/Windows/Linux + OS version):windows 7
  - Connected devices or roasting machine: phidget 1048
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-08-25:
Readings beyond 1000 should be recorded if no filter is set, but the LCDs were showing readings only up to 999 if decimals were enabled. This is now changed in v1.4 such that LCDs show readings beyond 999 (without decimals) in any case.

closed 2018-08-25 by MAKOMO

RoasterJ, 2018-08-28:
thanks a lot. I had figured out the filter part of it but not the decimal. You guys are awesome.