artisan-roaster-scope/artisan #242: Brambati

PRANTONY, 2018-08-26:
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version:
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

PRANTONY, 2018-08-26:
hi Everyone,I am new in this blog and i would like if anyone has experience to install artisan in Brambati coffee roasters.
Any Help?
Thanks

PRANTONY, 2018-10-29:
I ALREADY HAVE  A PHIGET SENSOR 4 INPUT INSTALL IN THE BRAMBATI KS 30 BECAUSE the previous roaster used it with another software. everything is already install i need just some help to configutare it

MAKOMO, 2020-09-25:
This is not a blog or a forum to help with individual configuration issues, but the issue tracker of the Artisan software development. Sorry. I'll close this.

closed 2020-09-25 by MAKOMO