artisan-roaster-scope/artisan #243: Control error

MountainRoaster, 2018-08-29:
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version:
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

[artisan-settings.zip](https://github.com/artisan-roaster-scope/artisan/files/2332038/artisan-settings.zip)

[435 Mt.Mornings Project 2.zip](https://github.com/artisan-roaster-scope/artisan/files/2332046/435.Mt.Mornings.Project.2.zip)
  File "artisanlib\main.py", line 11430, in <lambda>
  File "artisanlib\main.py", line 22631, in PIDcontrol
  File "artisanlib\main.py", line 43905, in __init__

--------------------------------------------------------------------------------
artisanlib\main.py::<lambda>:11430
	                   _ = False
	                self = <artisanlib.main.ApplicationWindow object at 0x00000188398EB678>
artisanlib\main.py::PIDcontrol:22631
	           modifiers = <PyQt5.QtCore.KeyboardModifiers object at 0x0000018842721358>
	                self = <artisanlib.main.ApplicationWindow object at 0x00000188398EB678>
artisanlib\main.py::__init__:43905
	   BTthermolabelnote = <PyQt5.QtWidgets.QLabel object at 0x00000188429B03A8>
	      button_readpid = <PyQt5.QtWidgets.QPushButton object at 0x00000188429A1288>
	  button_autotuneOFF = <PyQt5.QtWidgets.QPushButton object at 0x00000188429A11F8>
	   button_autotuneON = <PyQt5.QtWidgets.QPushButton object at 0x00000188429A1168>
	              dlabel = <PyQt5.QtWidgets.QLabel object at 0x000001884299ED38>
	              ilabel = <PyQt5.QtWidgets.QLabel object at 0x000001884299ECA8>
	              plabel = <PyQt5.QtWidgets.QLabel object at 0x000001884299EC18>
	            button_d = <PyQt5.QtWidgets.QPushButton object at 0x000001884299EB88>
	            button_i = <PyQt5.QtWidgets.QPushButton object at 0x000001884299EAF8>
	            button_p = <PyQt5.QtWidgets.QPushButton object at 0x000001884299EA68>
	          svwarning2 = <PyQt5.QtWidgets.QLabel object at 0x000001884299E8B8>
	          svwarning1 = <PyQt5.QtWidgets.QLabel object at 0x000001884299E828>
	     tab2getsvbutton = <PyQt5.QtWidgets.QPushButton object at 0x000001884299E708>
	        tab2svbutton = <PyQt5.QtWidgets.QPushButton object at 0x000001884299E558>
	         button_exit = <PyQt5.QtWidgets.QPushButton object at 0x000001884299E4C8>
	   button_standbyOFF = <PyQt5.QtWidgets.QPushButton object at 0x000001884299E438>
	    button_standbyON = <PyQt5.QtWidgets.QPushButton object at 0x000001884299E3A8>
	        button_rsoff = <PyQt5.QtWidgets.QPushButton object at 0x000001884299E318>
	         button_rson = <PyQt5.QtWidgets.QPushButton object at 0x000001884299E288>
	       button_getall = <PyQt5.QtWidgets.QPushButton object at 0x000001884299E1F8>
	        labelpattern = <PyQt5.QtWidgets.QLabel object at 0x0000018842987B88>
	              parent = <artisanlib.main.ApplicationWindow object at 0x00000188398EB678>
	                self = <artisanlib.main.PXRpidDlgControl object at 0x0000018842987A68>
	           __class__ = <class 'artisanlib.main.PXRpidDlgControl'>

MAKOMO, 2018-08-29:
Just fixed this typo. Thanks for reporting!

closed 2018-08-29 by MAKOMO

MountainRoaster, 2018-08-29:
Mark

Where do I find the fix ( non computer friendly person)

Gene

MAKOMO, 2018-08-30:
You can find pre-builds of the upcoming Artisan v1.4 on bintray: https://bintray.com/artisan/artisan-artifacts/artisan/master#files

Alternatively, you wait until the final Artisan v1.4 including the patch is realeased.