artisan-roaster-scope/artisan #244: Support older versions of macOS

cespare, 2018-09-16:
I have just started using Artisan and I really appreciate such great open source software.

My Macbook is currently running El Capitan (10.11.5) so I needed to install Artisan 1.2, as that is the most recent release which supports my OS.

Is there a particular reason why Artisan requires macOS 10.13 starting with version 1.3? I couldn't find any information about this in the Artisan 1.3 release notes. If it's not too much trouble, I think it would be better to maintain compatibility with the last few versions of macOS. It seems a bit aggressive to require an OS version that is only 7 months old (as was the case with macOS 10.13 back in April when Artisan 1.3 was released).

MAKOMO, 2018-09-17:
The reason for the move to OS X 10.13 is kind of external to Artisan and has to do with its underlying libs. I put some effort in this and created a build of the upcoming Artisan v1.4 that should run on OS X 10.10 and newer. However I cannot test this and need you support here.

Could you please contact me @
https://artisan-roasterscope.blogspot.com/p/contact-me.html

cespare, 2018-09-17:
@MAKOMO thanks! That was very prompt. I contacted you via the form.

MAKOMO, 2018-10-03:
legacy build supporting OS X 10.10/11/12 added for v1.4

closed 2018-10-03 by MAKOMO

cespare, 2018-10-03:
Thanks so much for working on this @MAKOMO! I'm enjoying some coffee that I roasted with my preview copy of 1.4 on my old macbook right now :) :coffee:

Evermorekoffie, 2019-05-22:
Hello I'm new. and I was wondering which Artisan version runs on Mac OS 10.12.6.
The 1.6.2 version doesn't! And where can I find it?

MAKOMO, 2019-05-22:
Dear Evermorekoffie,

according to the release history

https://github.com/artisan-roaster-scope/artisan

the last Artisan version supporting OS X 10.12 is v1.2.

You find all releases by clicking "All Releases" on the main page of https://artisan-scope.org/
which leads you to

https://github.com/artisan-roaster-scope/artisan/releases

M