artisan-roaster-scope/artisan #245: Cmd-S on Mac

InvisibleOrdinary, 2018-09-26:
## Expected Behavior
Cmd-S is save file

## Actual Behavior
Cmd-S works to both toggle sliders on & off, and also save file. This is confusing to have the same command work for different actions. toggle sliders should be a different cmd- shortcut, particularly because the shortcuts seem to be active at different times… (?)

This was logged as an issue in a prior issue, and was supposed to be included in 1.3.1 release, however it doesn't seem that this was included. Is it still on the roadmap?

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version: Artisan 1.3.1 (1) 
  - Artisan Build (number in brackets shown in the about box): (1)
  - Platform (Mac/Windows/Linux + OS version): Mac OS 10.14
  - Connected devices or roasting machine: Phidgets 1048 w/ 2 probes
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-09-27:
CMD-S is Save file, CMD-SHIFT-S is SaveAs (added in v1.3.1, Issue #194 ) and just S shows/hides the sliders. There is no overlap!

closed 2018-09-27 by MAKOMO

InvisibleOrdinary, 2018-10-03:
Hi Marko, 

There is actually overlap here. I am able to reproduce consistently. Which is why I keep submitting this…

Best I can identify is that the roast has to be in an OFF state for cmd-s to trigger save file. Otherwise, (in ON or START state) it is overridden by toggle sliders.

The same holds true for cmd ⇧ s toggling the sliders instead of save as

Please verify that you can reproduce this. 

Artisan Version: Artisan 1.3.1 (1)
Artisan Build (number in brackets shown in the about box): (1)
Platform (Mac/Windows/Linux + OS version): Mac OS 10.14

MAKOMO, 2018-10-03:
In ON and OFF state you cannot save a file (mostly because there is no complete file to be saved yet) an the Save menu items are therefore disabled, thus there is no overlap.