artisan-roaster-scope/artisan #250: Keep On and Phases LCDs All; Setting Persistence in 1.4.1

arkenamun, 2018-10-05:
Hi Marko, 

Some excellent new features, thanks very much to you and the team!

Just an observation, not an issue for me, that setting sampling "keep on" persisted after ticking that box then re-loading an old configuration file. (The 'all phases' check box also did this, but not the individual phase LCD options).

Thought it was worth pointing out to you in case it is not intended behaviour. Of course, please close if not considered an issue.

Attached is an old simple .aset file, where "keep on" has not been set. If you enable keep on then load these settings, you can see that keep-on is still active. So effectively the performance of the previous settings file has changed , and let's say if START were rule based after ON, then roasting could start unexpectedly.

Regards,

Brad

arkenamun, 2018-10-05:
[artisan-settings-1.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/2449040/artisan-settings-1.aset.txt)

MAKOMO, 2018-10-05:
Oh, you have already access to Artisan v1.4.1;) Wow!

This is the expected behavior. Older Artisan versions know nothing on new features/settings introduced in later versions. Thus application settings generated by older versions cannot modify features of newer app versions on loading into a newer app version.

Effectively the performance of the previous setting file has changed as you ticked that "keep on" flag before loading the older settings file (which could not change that flag back to its default as on its creation it couldn't know that this flag would be introduced later).

closed 2018-10-05 by MAKOMO