artisan-roaster-scope/artisan #253: WebLCDs cannot be enabled

homtherobot, 2018-10-05:
## Expected Behavior
WebLCDs functionality is enabled by ticking the "Port: 8080" checkbox.

## Actual Behavior
WebLCDs cannot be enabled and the checkbox is immediately cleared when ticked.

## Steps to Reproduce the Problem

  1. Go to Config >> Curves > UI tab
  2. Tick the "Port" checkbox in WebLCDs panel 

## Specifications

  - Artisan Version: 1.4.0
  - Artisan Build (number in brackets shown in the about box): b3b5e991
  - Platform (Mac/Windows/Linux + OS version): Windows
  - Connected devices or roasting machine: Center 304
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.
[artisan-settings.zip](https://github.com/artisan-roaster-scope/artisan/files/2450936/artisan-settings.zip)

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

arkenamun, 2018-10-08:
... yes just spotted that too, ran to the loo, couldn't connect, ran back. Now going to the loo between roasts as a work-around :-)

MAKOMO, 2018-10-09:
Fixed. As usual beta builds including this fix are available on BinTray https://bintray.com/artisan/artisan-artifacts/artisan/master#files

closed 2018-10-09 by MAKOMO