artisan-roaster-scope/artisan #254: Curve graphing stops after about 60 seconds

covabrelli, 2018-10-08:
I expect the temperature data to be visible during the roast for my ET curve and BT curve.

On v1.4 for Windows, my temperature curves disappear after about 60 seconds.  I have two temperature curves, ET & BT.  The data for these shows in the LCD windows but the graphing of this data disappears after about 60s into the profile.  If I click on an configuration option from one of the drop downs during the roast, the data curve to that point will appear but the ongoing data will be invisible.

## Steps to Reproduce the Problem

  1. Start roast, on & start
  2. Charge button
  3. 60s later, graphing of ET & BT disappears

## Specifications

  - 1.4.0
  -  (b3b5e991)
  - Windows-10-10.0.17134-SP0
  - Amprobe TMD-56
  
[Amprobe.artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/2457945/Amprobe.artisan-settings.txt)

Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-10-09:
Thanks for reporting! Could you please attach a profile recorded with that effect (even just a partial one up to 100s after CHARGE) for easier debugging?

covabrelli, 2018-10-09:
Marko,

I had to switch back to v1.3.1 to do my roasting yesterday.  I did a
complete uninstall of v1.4.  I didn't save any of the affected profiles.

Lou
Covabrelli Coffee
"Organic Coffee Delivered Fresh"
http://www.covabrelli.com
360.480.7003

covabrelli, 2018-10-09:
Marko,

After finishing the roast, all the data and all the graphs would be shown
correctly.  The disappearing graphs would happen during the recording of
the roast.

Lou
Covabrelli Coffee
"Organic Coffee Delivered Fresh"
http://www.covabrelli.com
360.480.7003

MAKOMO, 2018-10-09:
I tested your settings on my simulator, but couldn't detect anything odd (just wondering what you defined those quantifiers for). Without a profile it will be impossible to investigate and fix this one. Could you attach at least one of the profiles recorded with v1.3 under those application settings?

I understood that the curve only disappears during recording and the actual data is indeed recorded just not displayed. After the roast it is all nicely there. There must be something going wrong with the rendering during roasting, but without data it will be tough to find that needle.

MAKOMO, 2018-10-09:
Dear Lou, I might have found the cause of the disappearing curves and fix it. Could you please test if the current beta builds of v1.5 (that might be released as v1.4.1 bug fix release) on BinTray?

https://bintray.com/artisan/artisan-artifacts/artisan/master#files

Thanks,
Marko

covabrelli, 2018-10-09:
Marko,

Will do.

Thank you!

Lou
Covabrelli Coffee
"Organic Coffee Delivered Fresh"
http://www.covabrelli.com
360.480.7003

covabrelli, 2018-10-09:
Marko,

You fixed it.  Great job!

Lou
Covabrelli Coffee
"Organic Coffee Delivered Fresh"
http://www.covabrelli.com
360.480.7003

MAKOMO, 2018-10-09:
Thanks for confirmation!

closed 2018-10-09 by MAKOMO