artisan-roaster-scope/artisan #260: Multiple execution of Artisan

poundy, 2018-10-10:
## Expected Behavior
Be able to open multiple instances of Artisan executable at a time

## Actual Behavior
Since v1.4, this breaking change stops opening anything other than a single instance.

This is a large change, that seems to have no real justification for the change, and it's quite impactful to some as has been discussed on the list. 

As a separate "process" issue, it might be good if this kind of change in the future was noted as an issue here and was then open for discussion prior to it being implemented?  It feels like this wasn't a widely considered change 

## Steps to Reproduce the Problem

  1.  Open Artisan
  2. Try to open Artisan again
  3. 

## Specifications

  - Artisan Version: v1.4
  - Artisan Build (number in brackets shown in the about box): b3b5e991
  - Platform (Mac/Windows/Linux + OS version): Windows 10 (1809)
  - Connected devices or roasting machine: TC4-compat but issue is independent
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.  (unnecessary as this is a recognised change noted in the changelog)

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-10-12:
Added back the possibility to run a second instance of Artisan, but in restricted viewer-only mode

closed 2018-10-12 by MAKOMO

MAKOMO, 2018-10-13:
Dear  poundy

Please try out the current v1.5 builds on BinTray that feature an ArtisanViewer mode for the second started instance (besides fixing many bugs shown of in v1.4) and report if this helps.

https://bintray.com/artisan/artisan-artifacts/artisan/master#files