artisan-roaster-scope/artisan #262: Two extra lines drawn - indicate fan and heat levels

RickGroszkiewicz, 2018-10-14:
## Expected Behavior

## Actual Behavior
I noticed something different with artisan 1.4 that I have not seen documented anywhere. For roasts with my new Hottop B-2K+, there are 

two lines drawn that indicate fan and heat levels. They show up between 0 and 100 degrees F (on the left axis). These lines do not appear when I roast on my old Hottop B-2K.

## Steps to Reproduce the Problem

  1. Start roast with Hottop B-2K+ and artisan 1.4
  2. Use Config / Axes to change display of temperature axis to start at zero
  3. Now extra lines will be visible

## Specifications

  - Artisan Version: 1.4
  - Artisan Build (number in brackets shown in the about box): b3b5e991
  - Platform (Mac/Windows/Linux + OS version): Linux Mint
  - Connected devices or roasting machine: Hottop B-2K+
  
[Yemen Al-Haymah Crown Jewel 2017_18-10-13_1513.png.txt](https://github.com/artisan-roaster-scope/artisan/files/2476695/Yemen.Al-Haymah.Crown.Jewel.2017_18-10-13_1513.png.txt)

[artisan-settings-Linux-mint-Hottop 2K+.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/2476691/artisan-settings-Linux-mint-Hottop.2K%2B.aset.txt)

[Yemen Al-Haymah Crown Jewel 2017_18-10-13_1513.alog.txt](https://github.com/artisan-roaster-scope/artisan/files/2476689/Yemen.Al-Haymah.Crown.Jewel.2017_18-10-13_1513.alog.txt)

[Ethiopia Galena Abaya Yirgacheffe 2016_18-10-13_1441.alog.txt](https://github.com/artisan-roaster-scope/artisan/files/2476690/Ethiopia.Galena.Abaya.Yirgacheffe.2016_18-10-13_1441.alog.txt)

RickGroszkiewicz, 2018-10-14:
[image]

MAKOMO, 2018-10-15:
Rick, why ever, the two channels of the first extra device in your attached configuration is set to show the corresponding curves which record heater and fan values as received from the machine. The default Hottop 2k+ machine setting hides them. Just untick those two flags as below should resolve this.

[image]

Please let me know if this resolves your issue.

Note that some issues in v1.4 have been detected regarding the Linux builds and the Hottop communication that are now resolved in v1.5 to be released this week.

RickGroszkiewicz, 2018-10-15:
Thanks, Marco! I edited my settings via Config / Device / Extra Devices to un-check the boxes for curve1 and curve2. Now the extra lines are gone. 
I did not see those check boxes before. I had to use the horizontal scroll bar since they were not in view.

closed 2018-10-15 by RickGroszkiewicz