artisan-roaster-scope/artisan #263: Alarm Start by The Button or Slider

reizao, 2018-10-15:
## Expected Behavior
Can I click on one button or move an slider and trigger an alarm?
For example, if the Fan is set to 25% the alarm starts?

## Actual Behavior

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version: 1.4 
  - Artisan Build (number in brackets shown in the about box): b3b5e991
  - Platform (Mac/Windows/Linux + OS version): Linux
  - Connected devices or roasting machine: Phidgets 1048 / 1051

  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-10-15:
What for?

reizao, 2018-10-15:
Because instead of click on 3 buttons when I start the caramelization (25%
Fan, 10% heat, and Start of Caramelization) e Can click only in one button
and trigger one alarm to warn me when I have 2 minutes on caramelization.
Another request is to mark cusstom phases, like chemical phase 2.

Sorry about my poor english and Thanks,

Rodrigo Ramos

MAKOMO, 2018-10-16:
It does not makes any sense to me to click a button to get an alarm!? Alarm actions are triggered by time and/or temperature conditions, while button actions are triggered by user input. You can define an alarm that warns you with a popup 2min after FCs. You can another alarm that triggers directly after FCs (set the alarm time to 00:01) is set via the default FCs button and have this to trigger an "Event Button". You can define a custom "Event Button" to trigger 2 other custom event buttons via the "Multiple Event" action and those two other buttons you set to put the Fan to 25% and the heat to 10%. Note that you can hide all those 3 buttons. Read about alarms and events on the Artisan blog https://artisan-roasterscope.blogspot.com/ for details (you might want to use the link on its Overview page or the its search field).

Note that this tracker is not intended to serve as user support, but to report bugs or feature requests. Please use the public Artisan mailing list to ask for support.

closed 2018-10-16 by MAKOMO

reizao, 2018-10-17:
Dear Marko,

I think I not explained right. Some authors use sensory triggers to mark
phases of the roast instead of temperature or time. For example development
starts at vanilla smell instead of FC. And some procedures like
caramelization of sugars starts when the bread dough smell disappear. My
suggestion is customize these phases in a roast.

Thanks,

Rodrigo Ramos