artisan-roaster-scope/artisan #269: Unusual Screen Behavior

musicphanmike, 2018-10-18:
## Expected Behavior
Normal operation - screen should show tracking of BT/ET temperature as well as display on the right side of screen

## Actual Behavior
I've experienced two scenarios:
1) I typically press start while my roaster is in the last few seconds of temp stabilization.   Generally this would start the program as normal and start tracking and displaying temp.   Several times the screen would display the temps drop out to zero (or at least create vertical lines) - see Drop 1 pic.   There did not appear to be any consistent behavior as to when this would occur.  When I select Stop to restart the program I discovered it was tracking temps correctly (see Drop 1 Start pic) however it wasn't displaying on-screen correctly.   
[image]
[image]

2) I would start a roast and the information on the right hand side would simply disappear.  It will typically disappear just after I press the charge button however it does not appear to be connected to that function.  I would start the roast > select charge and the right screen would simply disappear.  The pic attached (loss of right pic) is showing the screen behavior.  This occurs about 20% of the time.. I can't seem to replicate any behavior that causes these issues.

[image]

## Steps to Reproduce the Problem

  1.  Unable to find anything consistant to understand the issue.   I've experiences similar scenarios on both my production roaster USRC 12K > Phidget > USB into Mac as well as my sample roaster - Quest M3 > Amprobe TMD-56 > USB to Mac

## Specifications

  - Artisan Version: 1.4.0
  - Artisan Build (number in brackets shown in the about box): b3b5e99
  - Platform (Mac/Windows/Linux + OS version): Mac High Sierra 10.13.6
  - Connected devices or roasting machine: USRC 12k > Phidget > Mac and Quest M3 > Amprobe TMD-56 > Mac
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.
[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/2493038/artisan-settings.txt)

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-10-18:
The first scenario is due to a bug introduced in v1.4, which is already solved by the just released v1.5. Please update to this version. The second scenario (missing LCDs on the right side) is the intended behavior. At any time you can show hide certain elements (incl. those LCDs named "Readings") via the View menu. Artisan remembers your selection per mode (OFF, ON, START). Thus if you miss any of display elements, just enable them by ticking them in the view menu.

closed 2018-10-18 by MAKOMO

musicphanmike, 2018-10-18:
Thanks Marko - I will upgrade to V1.5 for the first issue.   Was this change to the Views (Off/On/Start) a new function in V1.4?  I have never inconsistent screen views like I'm expereincing.

MAKOMO, 2018-10-18:
The view menu was introduced in v1.3.0