artisan-roaster-scope/artisan #273: Windows 7, install w/no errors, will not run

alansusee, 2018-10-20:
## Expected Behavior. After install expect Artisan to start 

## Actual Behavior. Attempts to open program result in nothing happening, no errors, nothing.  What am I missing?

## Steps to Reproduce the Problem

  1. Install on fresh build win7
  2. Attempt to open
  3. Will not start, no process running, no errors

## Specifications

  - Artisan Version: Tried 1.5, 1.4, 1.3 all same
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):  Win 7 fresh install 
  - Connected devices or roasting machine: phidgets 1080
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-10-20:
Last version supporting Windows XP/7 is Artisan v1.1.0. Sorry.

closed 2018-10-20 by MAKOMO

alansusee, 2018-10-20:
Rats! Thanks.  I was trying to put an older PC to use as a dedicated
Artisan machine.  Back to Ubuntu on that PC I guess.

Thanks again for a great product.