artisan-roaster-scope/artisan #274: Combo not showing description info for sliders

InvisibleOrdinary, 2018-10-22:
## Expected Behavior

Combo to show the value for each slider at the node when combo is checked

## Actual Behavior

only the node is shown: see screen shot

[image]

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version: 1.5
  - Artisan Build (number in brackets shown in the about box): fc6ebda
  - Platform (Mac/Windows/Linux + OS version): Mac OS 10.14
  - Connected devices or roasting machine: n/a
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.

[181021_artisan-settings.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/2500106/181021_artisan-settings.aset.zip)

Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-10-22:
Could you please attach this profile for testing this. Might be a bug.

InvisibleOrdinary, 2018-10-24:
did you get the profile I emailed? Thought it would have shown in the thread…

Either way, I roasted again tonight and have some additional info: 

The combo displays correctly for the roast while it is recording, however, it fails to display in the background profile, or for older profiles… Not sure whether older means before combo is checked, or before V1.5…

I would expect it to display in the background profile as well

here are two profiles: one from tonight (Duromnina) and one from before v 1.5 (Gorgora)

[Profiles.zip](https://github.com/artisan-roaster-scope/artisan/files/2508967/Profiles.zip)

MAKOMO, 2019-05-31:
Added in de0aee56584a7a29d1ac19180deba9ad19b3b04e

closed 2019-05-31 by MAKOMO