artisan-roaster-scope/artisan #275: CMD-N & menu New crash app

InvisibleOrdinary, 2018-10-24:
## Expected Behavior

either selecting from the menu or CMD-N would save current roast and begin new roast

## Actual Behavior

it just hangs up, requiring a force-quit, and restart for both avenues to a new roast

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version: 1.5
  - Artisan Build (number in brackets shown in the about box): fc6ebda
  - Platform (Mac/Windows/Linux + OS version): Mac OS 10.14
  - Connected devices or roasting machine: Phidgets 1048, 2 probes, Proaster 10K
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.

[181023_artisan-settings.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/2508973/181023_artisan-settings.aset.zip)

Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-10-24:
I cannot reproduce this one. What are the exact steps to reproduce this crash? I loaded your settings in Artisan v1.5 on my Mac and pressed CMD-N and it works as expected. Please give a bit more information. Thanks!

MAKOMO, 2018-10-24:
Found the cause of this hang. This is a feature interaction with the "Always ON" flag (Config >> Sampling Interval).  For now you can just turn that one off. I will fix this for the next version. Thanks for reporting. Good catch!

MAKOMO, 2018-10-24:
Fixed by Commit 23080851ad41f7fd3438d426fa8e4ccee77f457d

MAKOMO, 2018-10-24:
You can test this using the current builds on BinTray https://bintray.com/artisan/artisan-artifacts/artisan/master#files

closed 2018-10-24 by MAKOMO

InvisibleOrdinary, 2018-10-24:
Thanks - I will test this build next week… 

On the Proaster question… I did open that once, but it didn't immediately seem to work… though that might have been my hurry… I do know that the phidgets was installed aftermarket.

But also, I am leasing time on this machine, and so don't own it and there are others who also lease time but don't use software (hence the multiple lines out form phidgets)… 

however, I will look to see if there is a different connection next week… and if you'd send me that setting file, I will try that too. 

Cheers

Dave