artisan-roaster-scope/artisan #276: Modbus + PID problem

Shin-nn, 2018-10-26:
## Expected Behavior
Modbus + PID should work

## Actual Behavior
Modbus + PID not working -> SV is rising not according to set PID. When i configure PID (k,i,d) to zero, SV should stay zero. It does not, is is rising until it is maximum.

## Steps to Reproduce the Problem

  1. confugure modbus
  2. set PID (k,i,d) to 0 in Control/PID
  3. DV is rising 

## Specifications

  - Artisan Version: 1.5.0
  - Artisan Build (number in brackets shown in the about box): fc6ebda
  - Platform (Mac/Windows/Linux + OS version): Linux Ubuntu 18.04
  - Connected devices or roasting machine: Arduino based roaster

MAKOMO, 2018-10-26:
This report is unclear. Set Value (SV) is an input to the PID algorithm and is not changed by it, so it cannot rise. It remains unclear if you are using the Artisan software PID or an external PID running on your Arduino based roaster. Please attach your Artisan settings file and give more information on your setup and how to reproduce the issue.

MAKOMO, 2018-12-11:
Closed as communication stopped.

closed 2018-12-11 by MAKOMO