artisan-roaster-scope/artisan #278: Artisan for cacao roasters

ssantini, 2018-11-09:
## Expected Behavior
Profiles for cacao roasting.

## Actual Behavior
Only developed for coffee roasting. It woud be nice to adapt Artisan for cacao roasting, there´re some differences between coffee and cacao roasting.

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version:
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-11-09:
What's missing?

MAKOMO, 2018-12-11:
Closed as thread stalled

closed 2018-12-11 by MAKOMO