artisan-roaster-scope/artisan #279: Ending roast automatically closes current roast and opens new profile

digitalpriory, 2018-11-13:
## Expected Behavior

In previous versions, I would end roast, add anything I want to the profile, save, then click 'ON' to start a new profile.  

## Actual Behavior

No idea why the above process was changed - to save a few seconds or something for back-to-back roasts - but now I am forced to save the profile when clicking 'OFF', end the new profile that's automatically created, reopen the last profile, add anything I need (roasted weight etc), save and then start a new profile.   

## Steps to Reproduce the Problem

  1.  Compare the above with older versions of Artisan

## Specifications

  - Artisan Version: 1.5.0
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version): Mac
  - Connected devices or roasting machine: FZ-94
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-11-13:
Sounds to me that you ticked the new "Keep ON" option in the Config >> Sampling Interval dialog, described here: https://artisan-roasterscope.blogspot.com/2018/10/artisan-v140.html

MAKOMO, 2018-12-11:
Maybe just a configuration. Communication stalled.

closed 2018-12-11 by MAKOMO