artisan-roaster-scope/artisan #280: Unexpected error with Roast/Cup Profile

bigl00z3, 2018-11-16:
## Expected Behaviour
Open and edit "Roast/Cup Profile..."

## Actual Behavior
Trying to open Roast/Cup Profile triggers an unexpected error.
Error content:

An unhandled exception occurred. Please report the problem on Github:
https://github.com/artisan-roaster-scope/artisan/issues
When reporting this issue, please include your settings file (export 
via menu Help >> Save Settings) and the details below.

An entry has been written to the error log (menu Help >> Error).

Version: 1.5.0, revision: 0
2018-11-16, 19:21:36
--------------------------------------------------------------------------------
<class 'IndexError'>: 
list index out of range

  File "artisanlib/main.pyc", line 24239, in flavorchart
  File "artisanlib/main.pyc", line 34163, in __init__
  File "artisanlib/main.pyc", line 34263, in createFlavorTable

--------------------------------------------------------------------------------
artisanlib/main.pyc::flavorchart:24239
	                self = <artisanlib.main.ApplicationWindow object at 0x104f2b8b8>
artisanlib/main.pyc::__init__:34163
	        defaultlabel = <PyQt5.QtWidgets.QLabel object at 0x11967b288>
	            settings = <PyQt5.QtCore.QSettings object at 0x11967b0d8>
	            helpFlag = 65536
	               flags = <PyQt5.QtCore.WindowFlags object at 0x1193044a8>
	              parent = <artisanlib.main.ApplicationWindow object at 0x104f2b8b8>
	                self = <artisanlib.main.flavorDlg object at 0x106b428b8>
	           __class__ = <class 'artisanlib.main.flavorDlg'>
artisanlib/main.pyc::createFlavorTable:34263
	                 val = 5.0
	        valueSpinBox = <artisanlib.main.MyQDoubleSpinBox object at 0x115cf39d8>
	           labeledit = <PyQt5.QtWidgets.QLineEdit object at 0x115cf3948>
	                   i = 9
	            nflavors = 10
	                self = <artisanlib.main.flavorDlg object at 0x106b428b8>

## Steps to Reproduce the Problem

  1. Open "Roast/Cup Profile..." with the attached Roast document.alog (it does work fine with other files). The error pop up opens instead of Cup Profile
       Note: this is a roast with a coffee that ended up having a big phenolic sensory defect. Therefore the cupping results was set to very low.

[S317 Angola Serra N'Tandayala OFF Phenolic_18-11-15_2202.alog.txt](https://github.com/artisan-roaster-scope/artisan/files/2590476/S317.Angola.Serra.N.Tandayala.OFF.Phenolic_18-11-15_2202.alog.txt)

  2. 
  3. 

## Specifications

  - Artisan Version:Artisan 1.5.0
  - Artisan Build (number in brackets shown in the about box): (0)
  - Platform (Mac/Windows/Linux + OS version): Mac OS 10.11.6 & 10.13.6, Raspberry Pi/Raspbian (Stretch)
  - Connected devices or roasting machine: Phidgets (only on the Raspberry but not on the other Mac computers where I am editing the cupping)
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-11-16:
Oh, playing around with the cupping dialog I just found (and fixed) a non-terminating recursion.

MAKOMO, 2018-11-16:
I see that you have set Flavor to 20 and Acidity and Body to 50. I wonder how you entered those values as the controls should only allow values up to 10. Hm.

bigl00z3, 2018-11-16:
No idea how it went to 20 and 50. I just increased the values on the normal 0-10 scales. 
Could it be that if values are very low, it creates an issue? I think I moved Balance and Overall to 0.75 or something similar as the cup was totally out.

bigl00z3, 2018-11-16:
OK I just tried to change several values to below 1 on a profile, saved it, reopened it and the displayed values are not what I just entered before.
There is definitely something weird happening.

MAKOMO, 2018-11-16:
Found it in the code. On loading a profile there was a hack that tried to detect profiles generated with  v0.5 for values lower than 1 and lift them to a changed "interpretation" of later versions. That code had a bug that was responsible for both, the changed values and the difference in length between the list of values vs the list of labels that triggered your original error. Fixed that. Expect a new build soon. Don't use values lower than 1 for now until you have the new version installed. 

Thanks for reporting as always!

PS: I attached a fixed version of your original profile. You have to re-set the values again (in the range 1-10). Sorry for the troubles...

[S317.Angola.Serra.N.Tandayala.OFF.Phenolic_18-11-15_2202-fixed.alog.txt](https://github.com/artisan-roaster-scope/artisan/files/2590727/S317.Angola.Serra.N.Tandayala.OFF.Phenolic_18-11-15_2202-fixed.alog.txt)

bigl00z3, 2018-11-16:
Thanks Marko You are amazing!