artisan-roaster-scope/artisan #281: Artisan 1.5.0 for Mac buttons not recording

nkontchana, 2018-11-18:
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version:
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

nkontchana, 2018-11-18:
https://www.dropbox.com/s/rriu2l0yt8di48z/Tanzania%20New%20Software%20artisan-settings.aset?dl=0
https://www.dropbox.com/s/kt926tn0glrygfv/Tanzania%20New%20Software%20Installation_18-11-17_1244.alog?dl=0

Makomo I do apologise for not replying correctly. 

Artisan 1.5.0 (fc6ebda)
for Mac

MAKOMO, 2018-11-18:
I am now totally confused. Just checked your settings and your profile. Looks all fine. With your setting, Artisan v1.5 on my Mac records events as expected. Could you maybe share an image or short video sequence that demonstrates that issue?

[image]

nkontchana, 2018-11-19:
Makomo Thank You,

If I look on my PC at the same page as your last post i.e Roast Properties / Events my Time FT BT etc values are blank. Further none of my buttons or events appear on the roast profile graph. Would a screen shot of the same images you sent me (of my roast profile and roast properties / events) help.
 
Thank you Lawrence

nkontchana, 2018-11-19:
Makomo Hi,

I think I have resolved the problem. Unfortunately I don't know why. 
Could you please let me know why there is a notable delay pressing an event button and it actually plotting on the graph. (+- 7sec)

Thank you for all your assistance .

Lawrence

MAKOMO, 2018-11-19:
Events are only "processed" once per sampling interval. Most likely your sampling interval is 7 sec, quite high.

nkontchana, 2018-11-20:
Makomo

Thank you......

closed 2018-11-20 by MAKOMO