artisan-roaster-scope/artisan #282: Entering Full screen with LCD open, ignores LCD

InvisibleOrdinary, 2018-11-21:
## Expected Behavior

When entering full screen, with LCD up, the LCD is still visible.

## Actual Behavior

Enter full screen mode with LCD visible, LCD is placed on another screen. When I did it was placed on a space (screen) two screens away

## Steps to Reproduce the Problem

## Specifications

  - Artisan Version: 1.6 
  - Artisan Build (number in brackets shown in the about box): f92eec5
  - Platform (Mac/Windows/Linux + OS version): MacOS 10.14
  - Connected devices or roasting machine: Phidgets 1048 w/ 2 probes
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

[181120_artisan-settings.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/2602675/181120_artisan-settings.aset.zip)

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2019-01-23:
The Mac OS X shows only one window in full screen mode, thus it is not possible to have the main Artisan window AND the LCDs window open in full screen mode. Placing of other open windows of an app is totally up to the operating system and not under the control of an app. Am I overlooking something?

InvisibleOrdinary, 2019-01-23:
hmm… perhaps this is a train that's long left the station… and can't easily be done

What I am used to, and from a user's perspective what I expected to experience, is that when moving into full screen mode, you don't lose anything you are working with - so any open window in that app… A good reference example is adobe CC software… though I realize they have greater resources to work this out.

But… just to put some rationale out there… what I like with the LCD is the size, so that when glancing at the screen from the front of the roaster, it's the first thing that I see - which is what I want to see first… I wanted full screen to show this, and get rid of the extra clutter outside the windows without having to be nit-picky about the laggy window sizing.

For now, I will just use in standard view

thanks

MAKOMO, 2019-01-23:
This is not a matter of resources but a behavior of the operating system, Mac OS X in this case. Try to put Mail.app in full screen mode. It will only show the front most window. There is no hack around this. I do not know Adobes CC software, but it might be that what it is showing is several non-native windows within a native OS X window. In that respect the project really lacks resources to implement a full window system. I leave that to others. Assuming you do not want to help out here with code, I close this one for now. Sorry.

closed 2019-01-23 by MAKOMO

InvisibleOrdinary, 2019-01-23:
Sorry, wasn't saying you didn't understand what OSX was demanding… was just giving a good working example of what I was trying to describe… 

And not that I wouldn't, but that I can't write code, so can't really contribute in that way, my expertise lies on the user side. 

Cheers

Dave