artisan-roaster-scope/artisan #287: Alarms Call Program not sending "

reizao, 2018-12-19:
## Expected Behavior

Call program send " to external program in alarms

## Actual Behavior

Call Program doesn't send "

## Steps to Reproduce the Problem

  1. Call program espeak "Turning Point"
  2. Is sent espeak Turning Point to program
  3. It doesn't work

## Specifications

  - Artisan Version: 1.6.0
  - Artisan Build (number in brackets shown in the about box): c90fda38
  - Platform (Mac/Windows/Linux + OS version): Windows 10
  - Connected devices or roasting machine: Probatone 5
[king.zip](https://github.com/artisan-roaster-scope/artisan/files/2692834/king.zip)

  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

reizao, 2018-12-19:
In Raspberry is working nice!

MAKOMO, 2018-12-19:
I cannot reproduce this one. I installed the lastest espeak on my Windows 8 and added a simple alarm as shown below. That one is triggered and outputs the word by voice as expected. This one contains an accent character to test if this is the problem. It worked for me. This does not necessary rule out potential encoding problems. The call_program action is implemented differently on Windows than Unix and is very OS specific. Does a simple alarm like mine work especially if it is without accent characters?

[image]

reizao, 2018-12-19:
This way the voice says only the first word. Try c:\espeak\command_line\espeak -vpt "Atençao! Verificar Fase Química 2. Q2. Odor de Baunilha" and the espeak says only the word Atençao. not the complete Phrase

MAKOMO, 2018-12-20:
Got it now. Fixed by f25c0beb700d4d12735781f89f5c0ef36e5b7841.

closed 2018-12-20 by MAKOMO