artisan-roaster-scope/artisan #289: Phidgets 1135_0 (Precision Voltage Sensor +/- 30V DC) not working in Artisan 1.6.0 (OK in 1.5.0)

bigl00z3, 2018-12-27:
## Expected Behavior
Phidget 1135_0 connected to Phidget HUB0000_0 should display voltage signal in Artisan 1.6.0 as it is the case in 1.5.0.
Other fidgets are working fine (1048, HUM1000, PRE1000).

## Actual Behaviour
Only value "-1" is displayed in version 1.6.0

## Steps to Reproduce the Problem

  1. Install Artisan 1.6.0 on Raspberry Stretch
  2. Connect Phidget 1135_0 in port 0 of HUB0000_0
  3. Add Phidget 1135_0 as an extra device "Phidget HUB0000 IO 01"
  4. Start a roast, only "-1" is displayed were the voltage should be

## Specifications

  - Artisan Version: 1.6.0
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version): Raspbian Stretch
  - Connected devices or roasting machine:

The following devices are connected:
- Raspberry Pi Stretch
- Phidget 1048 (temperature) usb **OK**
- Phidget HUB00000_0 
- Phidget HUM1000 (temperature & humidity) connected to HUB0000_0 port 4  **OK**
- Phidget PRE1000 (Pressure) connected to HUB0000_0 port 4  **OK**

- Phidget 1135_0 connected to HUB0000_0 port 0  **NOT OK**
  
In Artisan v1.6.0 Raspian (artisan-linux-1.6.0_raspbian-stretch.deb), my Phidgets 1135_0 (Precision voltage sensor +/- 30 V DC) connected to a VINT HUB0000_0 (port 0) which is used in order to read my fan power is not working anymore.
However, my Phidget 1048 (temperature sensor) is still working fine at displaying the temperature same for the Ambient Phidgets.

I have tested:

- Downgrading to Artisan **v1.5.0 —> fixes the problem.**
- Reinstalling **1.6.0   —> recreates the problem**.
- Doing a **factory reset** for the artisan-settings **—> does not fix the problem**.

Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

[artisan-settings 1.6.0_problem 1135_0.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/2713244/artisan-settings.1.6.0_problem.1135_0.aset.txt)

[artisan-settings v1.5.0 working fine.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/2713268/artisan-settings.v1.5.0.working.fine.aset.txt)

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

bigl00z3, 2018-12-27:
Just did the same test on my Mac OS X (10.13.6). I get the same problem (and resolution). So it's not linked to Raspbian only:

- works in 1.5.0 (build fc6ebda)
- does not work in 1.6.0 + (build e8dfe2b)

MAKOMO, 2018-12-31:
Thanks Nico for reporting and testing also on Mac OS X. There was a typo in one of the latest commits which is fixed now. That latest commit also rewrites parts of the Phidgets driver in Artisan for faster attach (and might break other things). Please test carefully!

closed 2018-12-31 by MAKOMO

bigl00z3, 2019-01-01:
Great it works. However, there seem to be a sampling rate issue now. I will open a separate issue.