artisan-roaster-scope/artisan #290: Please add support for multiple channel of Phidget TMP1100 (VINT, isolated)

jmunkdk, 2018-12-29:
## Expected Behavior
The coffee roaster in question is big and collects noise, so an isolated TC input like Phidget TMP1100 (VINT, isolated) has to be used. Unfortunately Artisan only supports one input. Should be very easy to add support for more, since the inputs are already handled over the VINT Hub Phidget.

## Actual Behavior
Phidget TMP1100 (VINT, isolated) only supports one thermocouple channels

## Steps to Reproduce the Problem

  1. 
  2. 
  3. 

## Specifications

  - Artisan Version: Any
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2018-12-29:
Artisan already supports any number of Phidget VINT TMP1100 devices. They are attached in order of their serial number. However, as each device is offering only one channel, you have to setup an Artisan (extra) device for each TMP1100 where it returns the TC signal in the first Artisan channel and the device internal ambient temperature (used for internal calibration) as second Artisan channel. If you want to use two TMP1100 for ET and BT, you need to setup one extra channel of type TMP1100 and map its signal via a symbolic function onto the ET channel while the first TMP1100 of the main channel gets automatically assigned to BT.

closed 2018-12-29 by MAKOMO

jmunkdk, 2018-12-29:
Thanks Marko.

 

I didn't spot that method, but I am pleased it is already supported. I will order some more TMP1100 devices now.

 

Thanks for your help and a wonderful program.

 

Happy New Year,

 

Jens.