artisan-roaster-scope/artisan #293: error in feedback line but Artisan is working

benoitmathieu, 2019-01-04:
## Expected Behavior
no error in feedback line

## Actual Behavior
see annexed screenshot
[image]

## Steps to Reproduce the Problem

  1. artisan linked to CR3(r3) Shield (TC4 substitute) running aArtisan
  2. clock on "ON" to connect to Arduino/Shield
  3. error line arrives within a few seconds

## Specifications

  - Artisan Version: 1.2.0
  - Artisan Build (number in brackets shown in the about box): 3
  - Platform (Mac/Windows/Linux + OS version): Win7
  - Connected devices or roasting machine: CR3(r3) Shield (TC4 substitute) running aArtisan
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.
[image]
[artisan-settings_error_line.txt](https://github.com/artisan-roaster-scope/artisan/files/2728544/artisan-settings_error_line.txt)

benoitmathieu, 2019-01-04:
Issue still present in v1.5.0 (fc6ebda9), but referencing line 38818

Benoit

poundy, 2019-01-04:
Can you confirm whether this has ever worked without this error? 
(from my memory, I've not seen anyone else report issues on ArduinoTC4 before, so it could be your local config or device, not in Artisan)

benoitmathieu, 2019-01-05:
Hello,

The error comment has been there since the beginning.
I just mention it for cosmetic purpose as it does not hamper the software from working fine.
Possibly other users might have just live with it.

If there is any other log or config info that might help in fixing the message, just let me know.

Benoit

MAKOMO, 2019-01-08:
I am a bit puzzled what the actual error reported here is about. Artisan expects either no answer or a line starting with "#" on issuing the CHAN; command on first connect, otherwise this error is raised. With the standard aArtisan or aArtisanPID sketch this error is not raised as that request is correctly answered and that might be the reason nobody encountered this issue before. Could you please clarify if any action is required here?

benoitmathieu, 2019-01-08:
Hello, as I said, it is not blocking at all proper functioning of the
software.
All the serial output has been copied in the initial entry, so the source
of the error line should be in it, except if it is about a timing problem.
I should also add the adapted aArtisan in the arduino is streaming data in
chan 3 and 4.
I could try with a vanilla release/config of aArtisan to see if the error
is still there.
I will keep you updated.

Benoit

benoitmathieu, 2019-01-13:
Hello,

I have investigated this issue and the problem seems to come from the "slow" answer of my system (uno memory is quite packed, so arduino answer might take a bit longer than a second to answer to CHAN;12134 and send back the "# Active channels set to 1234"... output.
As a consequence, Artisan is issuing the error notified in the first message and resend another CHAN;1234 command. (see serial log)
The expected answer arrives (# Active channels set to 1234) and Artisan is happy and working, but the initial error message is not flushed from the main screen. 
So I think it is just a matter of message flush, the last message being displayed until an new one is displayed. If everything goes well, no new message is displayed before the end of the roast
As I said, only cosmetic but misleading.

Cheers,

Benoit

MAKOMO, 2019-01-21:
I close this. Error messages are never flushed from the message line. Always the lat message remains until overwritten by the next one, which is ok in my opinion.

closed 2019-01-21 by MAKOMO