artisan-roaster-scope/artisan #295: When open roast properties scope is shifted to the right

reizao, 2019-01-11:
## Expected Behavior
Scope remains the same

## Actual Behavior
Scope is shifted to the right

## Steps to Reproduce the Problem

  1. Charge the roaster
  2. On the charge open roast properties
  3. After closing this window scope is shifted to the right
[image]

## Specifications

  - Artisan Version: 1.6.0
  - Artisan Build (number in brackets shown in the about box): 7b160bad
  - Platform (Mac/Windows/Linux + OS version): Windows 10
  - Connected devices or roasting machine: Probatone 5 + Phidgets
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

reizao, 2019-02-05:
> Dear Rodrigo, I cannot duplicate this behavior on my system. Could you
> provide some more information that help me to reproduce this? It must have
> to do with an element drawn on the screen reserving the wrong space. Which
> element could it be?

Dear Marko,

I don't know what happen. Because when you load the configuration file
everything is OK again.

Yesterday I got another strange thing happening. When I started the roast
the font of HUD button is one, after I click on HUD button the font is
changed, and when I stop HUD the font remains changed. I attached some
screens.

I'm looking to translate m.a.s.l into m.a.n.m in portuguese but I can't
find it.

Thanks,

Rodrigo Ramos

MAKOMO, 2019-02-05:
No screenshots attached for that HUD font button "show". Could you attach them?

MAKOMO, 2019-02-05:
There were no translations generated for "m.a.s.l" I will add them and add this translation too.

MAKOMO, 2019-02-05:
Oh. No need for HUD screenshots. I see the issue and will fix this. Thanks for reporting!

reizao, 2019-02-12:
I got that error again.

Happens on the charge, when the roast properties screen is open.

I printed the screen.

[image: error graph.jpg]

MAKOMO, 2019-02-12:
Rodrigo, there is no screenshot attached here. So you were running the latest builds? Could you please attach also a screenshot of the about box of the Artisan version you encountered this. If I understand correctly, this happens while the Roast Properties dialog is open and a CHARGE happens (via autoCharge!?).

reizao, 2019-02-12:
Yes. Via auto charge.

MAKOMO, 2019-03-17:
Should be fixed.

closed 2019-03-17 by MAKOMO