artisan-roaster-scope/artisan #297: Artisan stops after drop

ibrubeer, 2019-01-20:
## Expected Behavior
- it should align the Charges, and keep displaying the graph

## Actual Behavior
about 15 seconds after dropping beans, Artisan stops

## Steps to Reproduce the Problem

  1. Select "Start"
  2. Drop beans
  3. 

## Specifications

  - Artisan Version: 1.5.0
  - Artisan Build (number in brackets shown in the about box): (fc6ebda9)
  - Platform (Mac/Windows/Linux + OS version): Windows 10
  - Connected devices or roasting machine: Aillio Bullet R1, V1

  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

[artisan-settings ERROR.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/2777068/artisan-settings.ERROR.aset.txt)
[EVB97_19-01-20_1548 Error.alog.txt](https://github.com/artisan-roaster-scope/artisan/files/2777072/EVB97_19-01-20_1548.Error.alog.txt)

ibrubeer, 2019-01-20:
[image]

RickGroszkiewicz, 2019-01-22:
I think you are confusing charge (load beans) with drop (eject beans).

ibrubeer, 2019-01-22:
My bad. I absolutely meant to say that it stops after Charge. Thank you.
Also, I rebooted my computer, and closed and reopened Artisan. Same result.

poundy, 2019-01-22:
I can't repro this on my install (and I've been using 1.5 for months). Has this ever worked for you?

ibrubeer, 2019-01-22:
Yes, been using it for months also.

RickGroszkiewicz, 2019-01-23:
The graph shows Charge, followed by Drop after a few seconds.

RickGroszkiewicz, 2019-01-23:
Do you have AutoDrop turned on in Config / Events / Config? If so, you should try un-checking that.

ibrubeer, 2019-01-23:
I do have that selected. It has worked for months, but I will try
unselecting that and see if that works. Will let you know...

Dennis

rpaulo, 2019-02-05:
Fixed in commit 0ea238c8edbdca5ac7a3b825ed16a3aab63310e4.  Please try the latest beta build from http://dl.bintray.com/artisan/artisan-artifacts/

closed 2019-02-05 by rpaulo

ibrubeer, 2019-02-17:
> Fixed in commit 0ea238c
> Please try the latest beta build from
> http://dl.bintray.com/artisan/artisan-artifacts/

This fixed it! Works great now. Thanks.