artisan-roaster-scope/artisan #300: Chinese Word?

LysanderChung, 2019-01-23:
I use Ubuntu , not Windows.
I set the Language as "traditional Chinese", but it can't show the right words in the graph.
It just shows "□□". (but under Windows, it's no problem.)
So, can you tell me what type of "traditional Chinese" that Artisan use?? (under Ubuntu system)

LysanderChung, 2019-01-23:
fonts-nanum is supported for Korea Language, not for Traditional Language.

I will try to install other fonts for Traditional Chinese.

And, I find the language setup files under /usr/share/artisan/translation/ , two files for zh_TW, but I can't open them by text editor.

MAKOMO, 2019-01-24:
Those files with extension .ts are XML text files and should also open with a text editor. You can use the Qt Linguist to load and edit those files.

LysanderChung, 2019-01-24:
where is the files with extension .ts ?? I don't know where they are.

I ever try other language of different country, I find that not only Traditional Chinese but also Simplified Chinese, Thai and several languages that also just show "□□" in "Roaster Scope" graph.

But under Windows system, Artisan can show the correct words in "Roaster Scope" graph.

LysanderChung, 2019-04-06:
I have updated Artisan to ver. 1.6.2. In the language setting, from top to bottom, there are 25 different languages. Only 1st, 3rd, 9th, 16th, 17th, 18th, 20th and 23th that can show the right words in that language in "Roaster Scope" graph, the others will show English instead.
     BUT I find only "Simplified Chinese" and "Traditional Chinese" that can NOT show the correct words or English instead,  just show "□□" in "Roaster Scope" graph. In Artisan ver 1.5.0, there are several languages have the same problem, just like Thai, Japan and so on. But in Artisan ver 1.6.2, the problem is fixed.
     I don't know if Traditional or Simplified Chinese are consist of "double char" in computer, not like other languages. Maybe computer language library or function can't handle it or compile it as default.

ihjmh, 2020-05-19:
i am so sorry to say this . on raspberry still got this problem, plz help me

MAKOMO, 2020-05-19:
It seems that Chinese fonts under Raspian running Qt apps (Artisan is a Qt app) are not well supported. This is an issue external to Artisan so I close this one here.

According to this

https://www.raspberrypi.org/forums/viewtopic.php?t=184904

installing the missing fonts under /usr/local/qt5pi/lib/fonts resolves this issue.

closed 2020-05-19 by MAKOMO