artisan-roaster-scope/artisan #303: Feature Request

InvisibleOrdinary, 2019-01-23:
## Desired Behavior

Option to add additional LCD data to the large LCD window. An additional 1-2 data points, writ large would be helpful during roasting… alternately, larger LCD of the right of the main window would work too

## Actual Behavior

only able to see time, BT, & ET

MAKOMO, 2019-09-02:
Added large Delta, PID and Extra LCDs in commit 362da4b8861a18e81655d00c01bba79fc4819b7a

closed 2019-09-02 by MAKOMO

InvisibleOrdinary, 2019-09-03:
Thank you

can the current beta build include a mac OS version? sometimes they do, sometimes they don't, but this one didn't

cheers

Dave

MAKOMO, 2019-09-03:
Dave, sorry for this. It seems that sometimes our automatic build system is overwriting already uploaded artifacts. There seems to be a race condition. I manually triggered a new build for Mac OS and now the build is online (until we trigger another build by another commit).