artisan-roaster-scope/artisan #306: libgcc_s.so.1 must be installed for pthread_cancel to work

bergera, 2019-01-26:
Hello there, long time user trying to update to the latest version.

## Expected Behavior
## Actual Behavior
I am using a Phidget 1048_0 4-input. When the ON button is clicked, Artisan crashes with the following console error:

```
libgcc_s.so.1 must be installed for pthread_cancel to work
Aborted (core dumped)
```

> I can provide a core dump or Apport crash file if that you'd find it helpful.

I built `libphidget22-1.0.0.20190116` from source and verified that it correctly reads from my connected 1048_0. The Phidgets TemperatureSensor Python example code works and produces correct readings.

libgcc_s.so.1 is installed:

```
$ locate libgcc_s.so.1
/lib/i386-linux-gnu/libgcc_s.so.1
/lib/x86_64-linux-gnu/libgcc_s.so.1
/snap/core/5897/lib/i386-linux-gnu/libgcc_s.so.1
/snap/core/5897/lib/x86_64-linux-gnu/libgcc_s.so.1
/snap/core/6034/lib/i386-linux-gnu/libgcc_s.so.1
/snap/core/6034/lib/x86_64-linux-gnu/libgcc_s.so.1
/snap/core/6130/lib/i386-linux-gnu/libgcc_s.so.1
/snap/core/6130/lib/x86_64-linux-gnu/libgcc_s.so.1
/snap/vlc/555/lib/x86_64-linux-gnu/libgcc_s.so.1
/snap/vlc/768/lib/x86_64-linux-gnu/libgcc_s.so.1
/snap/vlc/770/lib/x86_64-linux-gnu/libgcc_s.so.1
/usr/lib/gcc/x86_64-linux-gnu/6/libgcc_s.so.1
/usr/lib/gcc/x86_64-linux-gnu/7/libgcc_s.so.1
/usr/share/artisan/libgcc_s.so.1
```

Using `LD_PRELOAD=libgcc_s.so.1` has no effect. I also tried rebuilding libphidget22 using `LIBS="-lgcc_s"` and that had no effect.

I have also tried versions of libphidget22 as old as `libphidget22-1.0.0.20170711` and still encountered the same problem.

## Steps to Reproduce the Problem

  1. Select any Phidget from Device menu.
  2. Click ON

## Specifications

  - Artisan Version: 1.5.0
  - Artisan Build (number in brackets shown in the about box): fc6ebda
  - Platform (Mac/Windows/Linux + OS version): Ubuntu 18.04.1 64-bit
  - Connected devices or roasting machine: Phidget 1048_0 4-input
 
[artisan-settings.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/2777495/artisan-settings.aset.txt)

```
version = 1.5.0 (fc6ebda)
Architecture = ('64bit', 'ELF')
Libc = ('glibc', '2.3.4')
Linux = ('debian', 'buster/sid', '')
Machine = x86_64
Platform name = Linux-4.15.0-43-generic-x86_64-with-debian-buster-sid
Processor = x86_64
Python Branch = 
Python Build = ('default', 'Nov 28 2017 18:47:20')
Python Compiler = GCC 5.4.0 20160609
Python Implementation = CPython
Python Revision = 
Python version = 3.6.3
Release = 4.15.0-43-generic
System = Linux
Version = #46-Ubuntu SMP Thu Dec 6 14:45:28 UTC 2018
```

bergera, 2019-01-26:
I don't know why it didn't occur to me before, but running Artisan as root does not cause a crash. Still, it's not ideal, so I hope this issue is still helpful. A user would need to understand X to get this running, and either launch Artisan from the command line or edit the .desktop file to change the `Exec` entry.

```
$ xhost +local:
$ sudo artisan
```

MAKOMO, 2019-02-06:
I am not a linux expert, but did you setup the udev rules for the Phidgets as noted in the installation documentation https://www.phidgets.com/docs/OS_-_Linux ?

MAKOMO, 2019-05-31:
No further replay on this.

closed 2019-05-31 by MAKOMO