artisan-roaster-scope/artisan #309: Mark FC Automatically by observering ET RoR trough- the Scott Rao way

samtsai5212, 2019-02-01:
## Expected Behavior
[feature request]
According to [Scott Rao article](https://www.scottrao.com/blog/2018/11/25/how-to-use-cropster-to-know-exactly-when-first-crack-began), we are able to mark the precise FC by observing the point of ET RoR trough near FC. If we can teach the program to find the lowest point near FC, the FC can be automatically and precisely marked.

## Actual Behavior
Find the lowest point of ET RoR near FC and mark it as FC automatically.

## Steps to Reproduce the Problem
  0. Scenario: If the BT was higher than Y(eg. 180C), after X seconds(eg. 20 second) and the ET RoR was not  recording a new low point, the system mark retrospectively the lowest point of ET RoR as the FC.
  1. Use a stack to keep the low point of ET RoR. 
  2. Variable Y: Temperature degree near FC
  3. Variable X: time after Temp Y
During the roast, the stack should keep replaced by a new low point of ET RoR, utill after X second of  Y degree of temp(the BT tempaerature assign by the user).
 
## Specifications

  - Artisan Version:
  - Artisan Build (number in brackets shown in the about box):
  - Platform (Mac/Windows/Linux + OS version):
  - Connected devices or roasting machine:
  
Please attach your current Artisan settings file (as exported via menu Help >> Save Setings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

RickGroszkiewicz, 2019-02-01:
I took a look at a few of my roasts. This method does not always locate First Crack correctly on my Hottop.

poundy, 2019-02-01:
This method also would be very variable depending on noise floor and responsiveness of probes.

samtsai5212, 2019-02-02:
Indeed, this method depends much on a responsive ET probe. I am using a 1.5mm probe and phidget 1048. I followed his rule of holding the gas around FC +/- 45 seconds and I was able to see the trough of ET RoR most of the time.

arkenamun, 2019-03-11:
I see Rao's FC trough in my deltaET for my 3mm ET probe on my 15kg roaster. This result is very reliable, matches his thesis/graphs and very prominent. Unfortunately it is entirely bogus and does not exist (in my case). I say this as I also have a super fast 1mm ET probe. So the 1mm LCD/graph reacts much much faster from the previous gas change and gives a better view of reality, with no significant peaks of ET RoR during this period. So I am not sold on this being anything less than an artifact and indicative of probe lag. Of course, your results may be more encouraging, but please try in parallel with a very this ET thermocouple to confirm if it does really exist before relying on this.

Brad

closed 2019-03-15 by samtsai5212