artisan-roaster-scope/artisan #311: Alarms load with Roasts Presets

reizao, 2019-02-12:
Only a feature request.

That a alarms and buttons config file is linked with roast properties preset (drop down list)

MAKOMO, 2019-02-14:
Since Artisan v1.4 you can tick "Load from background" in the Alarms dialog (menu Config >> Alarms). If this feature is activated, alarms are loaded from an alog profile (note that each alog profile contains the alarm list active the moment it was saved) if loaded into the background either manual via the "Profile Background" dialog (menu Roast >> Background) or if loaded via a Recent Roast preset. Note that from Artisan v1.6 on, the currently loaded profile is linked to a Recent Roast entry on creation together with the other Roast Properties like beans, weight in, weight out, ...). On selecting one of those Recent Roasts entry, the linked profile is automatically loaded into the background and with it the alarms if the flag discussed above is ticked.

However, button definitions cannot be linked to Recent Roasts. However, Palettes feature (menu Config >> Events, tab Palettes) allows you to define 10 sets of buttons/sliders/quantifier definitions that can quickly be switch by pressing CMD-<n> keys to switch between different configurations. Would that help already?

Read on palettes at the end of this post: https://artisan-roasterscope.blogspot.com/2013/02/events-buttons-and-palettes.html

MAKOMO, 2019-03-04:
Stalled. I close this.

closed 2019-03-04 by MAKOMO